kcbam/dbus-huaweisun2000-pvinverter #8: PV inverter missing with the new venus UI

Kappa71, 2024-11-02:
With the new UI introduced with venus 3.50 the sun2000 pv inverter is missing, both in the devices list and in the pv inverter settings.
Only the total inverter production in the overview still works and all data on the VRM portal.
With the old UI (still present on venus 3.50) all works as before.
Just for info, I have also the dbus-opendtu (https://github.com/henne49/dbus-opendtu) script and this works well also with new UI.

DenkBrettl, 2024-11-02:
Thanks for reporting this! It's not surprising that this doesn't work, as modifying the V2 GUI is something I haven't looked into so far.
Documentation on how to modify gui-v2 seems sparse, at least I haven't been able to find it easily. I've opened a topic in the Victron community section to gather the knowledge in implementing this in gui-v2. At the same time, I have very limited time in the next weeks, so I can't commit on making this happen within a certain time frame. For the moment, please use the gui-v1 to configure the SUN2000 parameters. Stay tuned :-)

DenkBrettl, 2024-12-29:
Just to update: This is still not working and it seems like it won't work any time soon in GUIv2. The issue is that modifying GUIv2 at this point in time is only possible by shipping a new WASM binary that was built before. There are two issues with this, which I've documented in a new thread on the Victron community https://community.victronenergy.com/t/modifying-gui-v2/10074 and I don't think it's time well spent to work on this issue until the way forward is clear.

Kappa71, 2025-01-02:
Thanks for the update.
I agree that modifying the GUIv2 is an overkill.
Maybe it's better to switch back for a simpler configuration via file.
What I miss more than the ability to configure the script via GUI, which is typically used only one time, is to see the inverter data via the GUIV2.
Referring to the dbus-opendtu mentioned before I see my two microinverters in the device list (see the attached images), while the sun2000 is completely missing, but present on the VRM website and computed on the GUIV2 overview solar yield.
I don't think that to populate the device list the GUIV2 must be rebuild because I've never done anything like that for dbus-opendtu. BTW with the GUIV1 the sun2000 was present in the device list.

[image]
[image]
[image]

agetommy, 2025-11-27:
The v1 GUI did not work for me with the "modification enabled = on" settings. And in case i did not load the modifications, these Sun2000 are not loaded the plugin was not displayed in the v1 GUI. 
I think it's the better way to configure these plugin via a configfile, in case a GUI v2 integration ist not possible.

Kappa71, 2026-01-31:
But now on the overview and on the VRM I see only the production on L1, why?

[image]