kcbam/dbus-huaweisun2000-pvinverter #10: v3.14 -> v3.54: pvinterter isn't exported via mqtt anymore

trisomeyr, 2025-04-07:
Hi,

first of all thanks for your work to integrate the Inverter into the OS.
Ive run this Setup successfully over a year now on a cerboGX with FW Version v3.14. I now wanted to upgrade to the current v3.54 to use the new Dynamic ESS function.
Unfortunately the pvinverter is not exported via mqtt anymore. In the Remote Console the Huawei shows up and all values are read but it simply does not show up via mqtt.
Can you verify that for your installation? How to fix that?

Thank you very much!

Runzelinho, 2025-07-29:
Hey, I have the same issue as @trisomeyr. Since v3.55, there is no more data in Victron MQTT and the Huawei inverter is no longer displayed (settings -> pv inverter -> Huawei...) in the remote console (V1) -> so I have the use the config.py to reconfig the setup.

Runzelinho, 2026-02-01:
Hey, thanks for your help. After the update, the inverter was still not displayed. I had to uninstall everything and reinstall. Then the inverter was displayed. Unfortunately, now I get the following error message:

```
root@einstein:~# python /data/dbus-huaweisun2000-pvinverter/dbus-huaweisun2000-pvinverter.py
2026-02-01 09:27:03,10 root INFO VRM pvinverter instance: 1
2026-02-01 09:27:03,12 root INFO Settings: ModbusHost '192.168.XX.XX', ModbusPort '502', ModbusUnit '1'
2026-02-01 09:27:03,13 root INFO Settings: CustomName '', Position '0', UpdateTimeMS '1000'
2026-02-01 09:27:03,14 root INFO Settings: PowerCorrectionFactor '0.995'
INFO:sun2000_modbus.inverter:Successfully connected to inverter
2026-02-01 09:27:05,39 sun2000_modbus.inverter INFO Successfully connected to inverter
2026-02-01 09:27:05,73 root INFO Static data received: {'SN': 'XXX', 'ModelID': 473.0, 'Model': 'SUN2000-12KTL-M501076370', 'NumberOfPVStrings': 4.0, 'NumberOfMPPTrackers': 2.0}
2026-02-01 09:27:05,74 root INFO Starting up
2026-02-01 09:27:05,126 root INFO No meter service created, as use_meter is set to 0
2026-02-01 09:27:05,129 root CRITICAL Error at main
Traceback (most recent call last):
  File "/data/dbus-huaweisun2000-pvinverter/dbus-huaweisun2000-pvinverter.py", line 336, in main
    dbus_service['service'].register()
  File "/opt/victronenergy/dbus-systemcalc-py/ext/velib_python/vedbus.py", line 93, in register
    self._dbusname = dbus.service.BusName(self.name, self._dbusconn, do_not_queue=True)
                     ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.12/site-packages/dbus/service.py", line 145, in __new__
    raise NameExistsException(name)
dbus.exceptions.NameExistsException: Bus name already exists: com.victronenergy.pvinverter.sun2000
root@einstein:~# 2026-02-01 09:27:05,74 root INFO Starting up
  File "/data/dbus-huaweisun2000-pvinverter/dbus-huaweisun2000-pvinverter.py", line 336, in main
    dbus_service['service'].register()
  File "/opt/victronenergy/dbus-systemcalc-py/ext/velib_python/vedbus.py", line 93, in register
    self._dbusname = dbus.service.BusName(self.name, self._dbusconn, do_not_queue=True)
                     ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.12/site-packages/dbus/service.py", line 145, in __new__
-sh: 2026-02-01: command not found
    raise NameExistsException(name)
dbus.exceptions.NameExistsException: Bus name already exists: com.victronenergy.pvinverter.sun2000
root@einstein:~# 2026-02-01 09:27:05,126 root INFO No meter service created, as use_meter is set to 0
-sh: 2026-02-01: command not found
root@einstein:~# 2026-02-01 09:27:05,129 root CRITICAL Error at main
-sh: 2026-02-01: command not found
root@einstein:~# Traceback (most recent call last):
-sh: syntax error near unexpected token `most'
root@einstein:~#   File "/data/dbus-huaweisun2000-pvinverter/dbus-huaweisun2000-pvinverter.py", line 336, in main
-sh: File: command not found
root@einstein:~#     dbus_service['service'].register()
>   File "/opt/victronenergy/dbus-systemcalc-py/ext/velib_python/vedbus.py", line 93, in register
-sh: syntax error near unexpected token `File'
root@einstein:~#     self._dbusname = dbus.service.BusName(self.name, self._dbusconn, do_not_queue=True)
-sh: syntax error near unexpected token `('
root@einstein:~#                      ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
-sh: ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^: command not found
root@einstein:~#   File "/usr/lib/python3.12/site-packages/dbus/service.py", line 145, in __new__
-sh: File: command not found
root@einstein:~#     raise NameExistsException(name)
-sh: syntax error near unexpected token `('
root@einstein:~# dbus.exceptions.NameExistsException: Bus name already exists: com.victronenergy.pvinverter.sun2000
-sh: dbus.exceptions.NameExistsException:: command not found
```

All devices are running the latest software version. Do you have any advice?

closed 2026-02-06 by DenkBrettl